camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #1: Question about the pinout to the Particle

pilotdeveloper, 2018-08-11:
How are you connecting this to your particle board?

I've got one and am able to test it too :)

camcamfresh, 2018-08-11:
turn on half duplex mode on the particle board, connect the yellow data wire from scooter to the serial's TX line and make sure to connect the ground too! after that it should programming like any normal system.

pilotdeveloper, 2018-08-11:
No need for a pull-up resistor and connection to RX?

Also, I'm going to open a branch with some additional screen support.

camcamfresh, 2018-08-12:
That's a good question. I'm not too sure. To the best of my knowledge it is not necessary, but I might be wrong. I think the pull up/down resistor just prevents unwanted noise when the line is not in use like during a reset or something. Although since it uses half duplex, the data line might "reset" between transmit and receive. If that is the case, it might be a major problem.

As of yet I haven't had any problems receiving or transmitting data, but I have not switched between receiving and transmitting data while in the main loop. I'll find out soon if it is required. If it is a problem, I think I will connect a 10 ohm resistor between the rx and tx lines.

neo3k, 2018-08-22:
Hi. I'm trying to make the connection, but I can't get it to work. Can you show me a picture of how you did it?  Thank you very much

camcamfresh, 2018-08-31:
I'll add a section in the README about how I connected it.

camcamfresh, 2018-09-01:
I added a part hardware implementation, which explains how I have it wired (with a transistor) and how to wire it if you only need the data communication.

closed 2018-09-17 by camcamfresh