camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #2: Serial vs BLE Protocol

losnir, 2018-08-16:
Hi!
First, thanks for the well detailed and coherent information, it will be of great use to me (:

Just making sure I understand correctly - the Serial 1-Wire protocol is exactly the same as the Bluetooth protocol (reverse engineered by CamiAlfa)?

My goal is to completely replace the original motor controller & BMS, but keep the the head (BLE) unit, so it will still look and behave like a stock scooter . 😁

camcamfresh, 2018-08-31:
I'm not too sure about the Serial 1-Wire protocol, though I have heard of it. I am pretty sure that the 1-Wire communication protocol is somewhat different than half-duplex serial communication, which also uses one (data) wire. 

To the best of my understanding, the Bluetooth protocol reverse engineered by CamiAlfa (which has helped me a lot) is a packet protocol which could theoretically be sent via the 1-Wire protocol. 

What are you trying to accomplish by replacing the original motor controller/BMS?

losnir, 2018-09-05:
The reason I'm replacing the electronics is that I'm upgrading to a 60V version of the scooter, possibly a dual-drive in the future.

Maybe I'm mistaken, I thought the data bus between the BLE and DRV is 1-Wire - is it half-duplex serial?

In other words, the communication between BLE/DRV over serial uses the same BLE protocol (`|x55|xAA| L | D | T | c |...|ck0|ck1|`)?

camcamfresh, 2018-10-26:
Sorry for the late reply,
I have only had success with the half-duplex serial.
And the communication over serial uses the same BLE protocol.

sigmounte, 2018-10-26:
In my case , spying on the serial protocol ( not one wire , just TTL half duplex ) , i see protocol who look pretty the same , biggest different is the CRC , it only finish with FE or FF in my case.
I'm waiting also trying to make a full replacement controller , to enable the use of a dual motor with two original Motor/ESC/Battery , piloted by one arduino

camcamfresh, 2018-11-07:
Hmmm... for the CRC I'm thinking that the byte before the FE or FF is also included in the CRC. I'd have to see what data your receiving to tell you more.

As for replacing the motor controller, that seems very daunting. I'm not very good with understanding the electrical parts of the scooter (ha), but this is why I didn't use any resistors on my first prototype of the BLE controller. 

Replacing the actual motor controller would seem to require you monitor how the BLDC motor is powered since to the best of my knowledge the three big wires switch rapidly between hot and ground (while the other is off). For example, if you have three wire A B and C: A could be ground for one cycle (where is a cycle is not necessarily a full rotation of the wheel, it could be 1/12 of a rotation or something like that) while B is the hot. The next cycle could have C as ground and A as hot. So you would first have to figure out the correct order in which the three big wires power the motor. 

On top of all of that you would also have to monitor the position of the motor (using the 5 smaller wires) to ensure that you do not apply power to the wheel when it is not in the correct position. If you do not monitor the position of the wheel and somehow the wheel is in the wrong position, it would likely lead to the motor burning out or not functioning correctly. I think this becomes a lot more important when attempting to suddenly stop or perhaps speed up.

This youtube video may give you a better idea of what I think you would have to do: https://www.youtube.com/watch?v=W9IHEqlGG1s&t=515s

It might be easier to purchase a second factory motor controller and connect that to a different serial port on the Particle Electron. This may or may not require another battery and would possibly require you to flash the motor controller to remove both the program defined speed limit and the hard speed limit on each motor controller (as each motor controller is not only programmed to not exceed specified speed limit but also to not exceed the physical limitations of the motor).