camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #3: How to activate Scooter

DoidoYo, 2018-09-10:


camcamfresh, 2018-10-26:
So I haven't tried enabling the password on the scooter, though I do remember seeing it on the MiHome app. I've tested the break and throttle information and as long as the scooter is on, the BLE controller sends the values, however if the scooter is locked the motor controller will not activate the motor (though it still receives the values). I would avoid using the word activate since the motor controller does need to be activated once to begin working (I'm not sure how this works but some guys in Russia activate the control boards for like $35 I think, if I find the link again I'll tag it on this thread). Now once it the motor control is activated, it will respond to commands sent by the BLE board. It doesn't matter if the motor controller loses power it will stay activated.  When I began experimenting with the communication data I literally copied the exact packets I was monitoring between the factory BLE board and the motor controller and programmed my microchip to send those. As a result, I had to spin the scooter's wheel before turning the microcontroller on (began communicating with the motor controller) but I was able to successfully turn on the motor although it was only at one speed.

jacknab, 2018-11-02:
hi,
I actually been looking for details on how to make my own BLE board for my scooter, I came across your project here on Github - which totally looks like an exciting project to play around with, I didn't finish reading the description/deals.. I spontaneously went for it.. and without any hesitation purchased the Particle Electron cellular kit.  
*Question now:  does the scooter still need the BLE board?
I have 2 M365 scooters, one is fully functioning, my second scooter I actually found in this very cluttered thrift store, sticker price was $10.00   -  however after much troubleshooting I have discovered the ble controller board was fried.

camcamfresh you mentioned you programmed your own microcrontroller could I ask what type of components you used?  Was it something Arduino like?

camcamfresh, 2018-11-07:
No it does not need the BLE board. I did purchase a BLE board, but that was strictly for learning about the messages it sends to the scooter's motor controller (that is where I learned about the information request packet structure; it's in the escControl function of firmware.ino).

I completely eliminated the BLE board from the scooter and replaced it with the Particle Electron. It is very similar to Arduino (both use C++ if remember correctly), but it also has cellular capabilities. 

microcontroller = particle electron (I also sometimes use it to describe the motor controller and BMS circuit boards). Not sure if this is the correct terminology, but I've never taken a computer science class about any of this so ... that's what I call it for now.

jacknab, 2018-11-07:
awsome i have ordered the particle cellular board however the version that
had gps board was sold out, the board arrivea tomarrow.

anyway of taking aome photos of your project to see its design and wiring.

it be awesome if there was a way to make a replacement ble board that
functioned the same as the original,  i short circuted my ble board last
month when creating a dispay using the arduino mini. a replacement board
found on ebay cost me $141.00 there were many listings for the ble board
but every other listing showed that part out of stock so was forced to buy
the 141.00 -  and when it arrived it took me a whole 10 mins to actually
fry that new board :(  i disxonnected the battery before hand, but there
was still electric current stored in the wiring or it got a small charge
when i rolled the scooter out of my garrage for better lighting..  but the
board touched the handle bar and i head a pop and off to ebay i went again

jacknab, 2018-11-07:
how are u sending commands to the particle board?  i know over celullar but
did u create a simple web page to do this

camcamfresh, 2018-11-07:
I usually use the Particle IDE, but am working on a swift program using HTML Post method (check the Particle Documentation). I am having trouble completing the program as I do not own a mac.

jacknab, 2018-11-07:
thanks for the quick response, I see you have also updated your code, just downloaded it - I received my Electron Particle couple hours ago- however its just the Electron it doesn't have gps board, because the site was sold out. As soon as they get them in stock the site will email me and I will purchase it.

I connected the electron to the 4 wires, still need to connect to throttle, light and motor wires.

I tried writing your firmware file to the electron but since I dont have gps its giving me errors, so I am going through and commenting out the gps stuff for the moment.

Arduino: 1.8.7 (Windows 10), Board: "Arduino/Genuino Uno"

sketch_nov07a:1:14: error: expected constructor, destructor, or type conversion before '(' token

 SYSTEM_THREAD(ENABLED);

              ^

C:\Users\chadb\Downloads\Xiaomi-M365-BLE-Controller-Replacement-master (1)\Xiaomi-M365-BLE-Controller-Replacement-master\firmware\sketch_nov07a\sketch_nov07a.ino: In function 'void setup()':

sketch_nov07a:7:5: error: 'Serial1' was not declared in this scope

     Serial1.begin(115200);

     ^

sketch_nov07a:10:5: error: 'Particle' was not declared in this scope

     Particle.function("CloudCommand", cloudCommand);

     ^

sketch_nov07a:19:13: error: 'D0' was not declared in this scope

     pinMode(D0, INPUT);//Power Monitor

             ^

sketch_nov07a:20:13: error: 'D1' was not declared in this scope

     pinMode(D1, OUTPUT);//Power

             ^

sketch_nov07a:21:13: error: 'D2' was not declared in this scope

     pinMode(D2, OUTPUT);//Buzer

             ^

sketch_nov07a:22:13: error: 'D3' was not declared in this scope

     pinMode(D3, OUTPUT);//Red (Lock)

             ^

sketch_nov07a:23:13: error: 'D4' was not declared in this scope

     pinMode(D4, OUTPUT);//Green (Unlock)

             ^

sketch_nov07a:24:13: error: 'D5' was not declared in this scope

     pinMode(D5, OUTPUT);//Blue (Connecting)

             ^

C:\Users\chadb\Downloads\Xiaomi-M365-BLE-Controller-Replacement-master (1)\Xiaomi-M365-BLE-Controller-Replacement-master\firmware\sketch_nov07a\sketch_nov07a.ino: In function 'void escControl()':

sketch_nov07a:68:5: error: 'Serial1' was not declared in this scope

     Serial1.write(motor, sizeof(motor)); delay(10);

[... 163 lines cut ...]

sketch_nov07a:268:9: error: 'Serial1' was not declared in this scope

         Serial1.write(cruiseon, sizeof(cruiseon));

         ^

sketch_nov07a:272:22: error: 'D1' was not declared in this scope

         digitalWrite(D1, HIGH);

                      ^

sketch_nov07a:278:14: error: 'D2' was not declared in this scope

         tone(D2, 20, 5000);

              ^

exit status 1
expected constructor, destructor, or type conversion before '(' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

closed 2019-04-05 by camcamfresh