camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #7: Brake and throttle are not working (back LED is blinking)

bnurbekov, 2018-10-26:
Hi,

Thank you so much for such a great work - it’s very useful. I was trying to run your code on my board. However, the brake and the throttle are not working in my case. Back LED is blinking 5 times (as if the brake and the throttle were disconnected from the Bluetooth board). I am wondering if I will need to send some additional messages to notify the main board that everything is connected properly. Please could you help me to resolve this issue? Thank you in advance!

camcamfresh, 2018-10-26:
It sounds like the motor controller isn't receiving any data at all. I think I remember, that the scooter would blink 5 times before I started communicating with my micro controller. As a result, I'm inclined to think that the motor controller is always listening for data from the BLE controller. If the BLE controller fails to send values for a few seconds it will activate the blinking until it receives the values. I will update the program on github later today to my current version.

bnurbekov, 2018-10-27:
@camcamfresh Thank you so much! After I flashed the latest code everything started working. Really appreciate your help.

closed 2018-10-27 by bnurbekov