camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #9: bit shifting instead of strings....

smartinick, 2018-11-18:
Not an issue, but as it seems you need to figure that out.... use it like that:

uint16_t crc = (motor1[2] + motor1[3] + motor1[4] + motor1[5] + motor1[6] + motor1[7] + motor1[8] + motor1[9] + motor1[10] + motor1[11] + motor1[12]) ^ 0xFFFF;
motor1[13] = (uint8_t)(crc&0xff);
motor1[14] = (uint8_t)((crc&0xff00)>>8);

maybe that safe you some time - your code templated saved my time (doing similar with the esp32).
Thanks for sharing!

for switching eco/normal mode you can look in my code, m365.cpp, method m365_sendcommand.

camcamfresh, 2018-12-10:
Thank you! My friend recommended that I do this, but my microchip broke before I could test if would work. I'll try replacing my code with this when I finish making my wire harness (probably next week).

smartinick, 2018-12-10:
you are welcome. thanks for starting this repo, saved me some time to
research myself :D
if you have any questions - ideas - i'm happy to give feedback/help!

closed 2019-07-05 by camcamfresh