camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #12: Control two ESCs from one stock (or other) Controller?

galstaf, 2018-12-14:
This project is in the same ballpark as what I am trying to do, so looking for some input from those who have way more knowledge than me. I have an M365 and that got me interested in the Bird scooters. I became a charger and briefly a mechanic (although that was frustrating), so now I just stick to the charging. 
Via my mechanic friends, I got a bunch of parts that Bird were dumping in the trash so I have a built a bench rig up with 2 motor wheels, 2 ESCs and 2 batteries packs.

I am not sure I need the cellular capabilities of the Particle Electron.. If anything I would like to use the stock BLE board and speed controlling rheostat switch on the stock scooter to control *both* ESCs and Both wheels to run at the same speed and power at the same time. 
I want to do this to build an two wheel drive scooter with a new custom body. 

Would it be as simple as splicing the TX wire off and running it to the second ESC? From what I understand from this project, this is how the wires go... 
ground (black) -- Yellow (TX) -- Green (40V pwr) -- Red (5V pwr)

It seems like there may be an issue with the BLE board getting communication from two ESCs, but perhaps not?

Can anyone help here? What would be the best way to address this and achieve my goal of one throttle control switch controlling two wheels at the same time?

BPitts2, 2018-12-14:
If you don't care to RX any info, you'll probably be fine. I haven't went
deep enough to have a consideration for handling messages from two machines
at the same time.

I don't see a reason that you couldn't at least have dual motor control
from one BLE replacement on one communication bus.

I also have two units at home. Now I'm really interested in making a 4
wheeled electric go-kart of some sort....

galstaf, 2018-12-14:
> If you don't care to RX any info, you'll probably be fine. I haven't went deep enough to have a consideration for handling messages from two machines at the same time. I don't see a reason that you [...]

Right? That sounds awesome too! 

However, it would be nice to get the RX info from at least one of the ESC units for speed and battery remaining information. I don't know how the BLE board would handle input from two ESCs at once. However it may be interesting to just try it and find out. 

Does anyone else have any thoughts on this?

galstaf, 2018-12-18:
PARTIAL SUCCESS! 

Just some feedback on using the stock BLE ... I tried controlling the two ESCs by splicing into the control wiring. With a little thought, obviously one cannot just connect the yellow, because the ESC requires the 40V and ground to tell it to turn on and off. It turns out  also it needs the 5V for something otherwise the motor will not activate... not entirely sure what the 5V is used for tho. 
So all 4 wires need to be connected to both of the ESC units. 

The flashed Botox firmware was then applied to both ESC units separately. Can anyone tell me if the firmware is applied at the BLE level or the ESC level or both?

ISSUES WITH THIS SOLUTION:

- You can connect the control software (I use M365 Tools), but it is obviously super confused as it is getting data from both of the ESC units, so the battery serial number and other data will randomly change.  It also doesn't show the speed correctly so it is pretty worthless. 

- The above issue is not a big deal, as in testing, both wheels worked together, but then randomly one of the ESCs will hang and the wheel will stop if the app is connected. So connecting the app will cause only drive wheel to eventually fail and stop. The ESC will stop responding to commands and has to be unplugged from the battery to reset it. 

- The ESCs do not respond to a power off command (long press shut off). It will go off for a second and then come back on again. The only way to effectively shut down the ESCs is to disconnect battery power
I think I am going to have create a physical power off switch to disconnect both the scooters from their batteries so I can shut it down. 

So it is kinda working, but not great. Still looking for suggestions on a better solution.

camcamfresh, 2019-02-19:
Firmware for like the hard speed limit and throttle control reside within the motor controller's firmware, but it's usually flashed through the microcontroller.

For your third bullet point, have you tried changing the timing of the poweroff command? also if you're using your own microcontroller, make sure the usb power isn't turning the scooter back on again.

galstaf, 2019-02-19:
It is a factory BLE (microcontroller as you call it)... the long press shutdown command doesn't seem to work with multiple ESCs connected. I think short of making a controller, the only resolution is to have a manual power switch to kill power to all the ESCs. 

On further testing, you are correct.. the firmware lives on the ESC, not the BLE. I can swap BLE boards and the same firmware is still on there. 

Thanks again for your input!

LeRaphael, 2019-09-09:
Hello
 i make Xiaomi M365 with 2 Motors and add Towing with 2 motor (Total road 4 Motor) I want use Arduino between the controllers and the display for dispatch information 

exemple : 
BLE - > Arduino -> ESC 1 
                           -> ESC 2 
                           -> ESC 3
                           -> ESC 4
 or 

BLE - > Arduino -> ESC 1 
                           -> ESC 2 
                           -> Arduino -> ESC 3
                                              -> ESC 4

i try with 2 ESC and one BLE with Data line connected with Y its working for Throttle and brake (down information), but for power on, turn on light, battery indicator and eco mode, and all ESC to BLE communication its not good (Up Information)
for this i want use Arduino for intercept up and down information and send or resend just what I need

You can help me ? you have finish your project ?

Sorry i'm french my English is not good

thomz0r, 2020-05-26:
funny , i was googling to find an arduino implentation and found that projet , i have in mind exactly the same goal .  i think best is to. drive natively both ESC and even add the abbility to enable it . 
i will try to port the dev on arduino only , no need GPS , not even BT .

bartoszwarszawa, 2020-06-03:
I think it would be the best to use boards like ESP with dual thread functionality as the timings on sending and reciving messages from drv controller is quite delicate and can't be disturbed by other processes. I already succeed in recreating the project with ESP32 and SIM808 GSM module instead of Proton board