camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #13: "Motor Control" vs "firmware"

Teklock, 2019-01-28:
Does anyone else have throttle issues with firmware.ino?  It's very strange but in short it don't work.  However when I build a firmware out of "Motor Control"  Everything works perfect!

BPitts2, 2019-01-28:
Have you reviewed the map, and also checked your grounds? I had to update
the map settings for my throttle, it was slightly higher than the existing
code. I also had an issue with a stray ground wire that was causing chaos.

Teklock, 2019-01-28:
I'm glad you asked this!  How do I find the limits for the throttle?

BPitts2, 2019-01-28:
I think I wrote a basic sketch to just read the input from the throttle and
print to serial. Then I took the min and max values and updated the map.

camcamfresh, 2019-02-03:
The factory halls sensors take an input 5V (you could use 3.3V), but the Electron’s pins are 5V tolerant.

BPitts2, 2019-02-03:
Camcamfresh is correct:

Notes: [1] FT = 5.0V tolerant pins. All pins except A3 and DAC are 5V
tolerant (when not in analog mode). If used as a 5V input the
pull-up/pull-down resistor must be disabled.

https://docs.particle.io/datasheets/cellular/electron-datasheet/

Teklock, 2019-02-04:
I understand the pins are 5v tolerant, but from what I've read on their forums ADC reference voltage comes from the 3.3v supply.  I guess for me to better understand, can you explain this line of code.

map(value, 685, 4095, 0x26, 0xC2)

I'm guessing the 4095 is the max (12bit) and 685 is the min?

BPitts2, 2019-02-04:
Yes, that is the analog input from min to max and mapping to the min and
max Hex codes. I didn't dig much deeper, but my assumption is that the
motor controller is expecting input from a min to max range. In order to
have a consistent throttle curve, the analog input is mapped to the min and
max Hex.

Anyone feel free to correct me if I'm wrong.

Teklock, 2019-02-04:
So why is the min 685 when there is no voltage?  Shouldn't it be 0?

Teklock, 2019-02-04:
Also why would the ESC expect a range from 38 to 194.  Where does that range come from?

camcamfresh, 2019-02-06:
> Yes, that is the analog input from min to max and mapping to the min and max Hex codes. I didn't dig much deeper, but my assumption is that the motor controller is expecting input from a min to max [...]
From what I remember, the motor controller's minimum acceptable value is 0x26. Anything below that, say 0x00, would cause the motor controller to think that the sensor isn't connected. I do think there is a maximum value for the motor controller too (e.g., it won't accept 0xFF), but I don't remember the exact value something like 0xF0. The above applies to both the brake and throttle sensor.

> Also why would the ESC expect a range from 38 to 194. Where does that range come from?
>
I'm not entirely sure on that. My best guess is that the minimum value ensures that the sensor is connected; for example, if the sensor wires were disconnected the microcontroller could read 0x0 or a smaller value (idk maybe from static from the air or case) like 0x15. As for the maximum value probably the same reason; for example, if the sensor wires were shorted, the sensor could send back 0xFF. This is just a guess though.

Teklock, 2019-02-06:
Makes sense, ty sir!

fyi... I have the throttle working very nicely on a teensy 3.1 with: 

map(control, 900, 4095, 0x26, 0xCC);

after you enable 12bit with:

analogReadRes(12);

closed 2019-02-19 by camcamfresh