camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #14: dependencies

mast3rshak3055, 2019-02-16:
What are all the dependencies you are using for this code? If you could supply a library name or download link.

mast3rshak3055, 2019-02-19:
#include "Serial4/Serial4.h"

m365-dash.ino:399:7: 'Serial4' was not declared in this scope

kyuhyong, 2019-02-28:
@camcamfresh 
I was trying to build the code using particle web IDE but could not figure out which is the one required.
I see "ParticleSoftSerial" exist but could not find Serial4 or similar.
Also for GPS, I found "Particle-GPS" but it has no member NMEA. 
Is NEO-GPS also works fine with the GPS onboard the Particle Electron?

camcamfresh, 2019-02-28:
Serial4 is only available when you select a Particle device that has a Serial4 port on it (this includes the Electron but not the Photon). So you'll receive error trying to compile Serial4 if the Photon is selected. As for the GPS library, I'm using Neo-GPS which has the member NMEAGPS.