camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #17: Cannot turn off the power

kyuhyong, 2019-03-01:
I did check the scooter is under control of particle electron for a while and have been testing.
Unfortunately I forgot to unplug the usb connector and send command to power off through cloud.
Then it won't turn off the scooter nor can command it even after usb disconnected.
Now 5V power is always on and cannot read tx data from either BMS or ESC.
I manually short 40V to Ground with no luck. 
I am desperate to know if there is a way to force reset the whole system. Please help.

kyuhyong, 2019-03-01:
I solved this issue by opening the bottom plate and unplug the power connector as seen here.
[image]

closed 2019-03-01 by kyuhyong

camcamfresh, 2019-03-04:
I had that same issue a few times with mine. Apparently the motor controller goes into an error state long enough to refuse communicating with the microcontroller. Do you know how to recreate the problem? I haven't had that problem in a while but want to be prepared for it.

nielswalther, 2020-01-18:
Hi  kyuhyong
Thanks Man, my son and I have just tried to install the Xiaomi pro dashbord, and it did not work, so we installed the old dashbord again, but now it was not possible to turn the scooter off and we could not connect to the app either. Your Fix solved the problem, so now I have a happy son and a working scooter again. THANKS!