camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #21: ES4 BLE Controller

mast3rshak3055, 2019-03-16:
If I shipped you a Segway ES4 could you adapt the code for that model as well?

camcamfresh, 2019-03-16:
Sounds interesting, probably could do, but not until I finish up a few things with my current project. I would also have a few concerns about ensuring it's not stolen.

mast3rshak3055, 2019-05-13:
I could buy one from Amazon and ship directly to you. Also I would love to contribute to the project! Cant wait to see the phone app. Do you have a PayPal or a way to accept a donation?

pilotdeveloper, 2019-05-13:
For what it's worth, I'd be happy to take a stab at it!

camcamfresh, 2019-05-16:
@mast3rshak3055  I've been pretty busy lately; I actually just started working on a CS degree, so I'm not sure how soon I would be able to get the chip to work with the ES4 (if I could at all). I'll know more of what my schedule looks like in about 2 weeks. I would assume it is the same (or very similar) protocol, have you tried connecting to a ES4 yet? 

As for the app, I actually haven't really made that many changes to the app since publishing the first version. Although, I did add support for Android Oreo and up. It really isn't that great. I had problems using Particle's Android SDK, so I had to do a lot of things manually. You can check it out on the play store it's called "Scooter Connect" (for lack of a better name). The pin configuration for the software on the app (that is flashed OTA to the M365 microcontroller) is:
```
int brakeInput = A5;
int throttleInput = A4;
int RLED = D5;
int GLED = D4;
int BLED = D3;
int buzzer = D2;
int headlight = D6;
int powerSwitch = D7;
```
The other wiring is the same. 
I also added a voltage divider for the brake and throttle inputs which fixed the throttle issues that I was experiencing (so at 3.3V [rather than 5V] the brake/throttle inputs will send the maximum values to the scooter).