camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #22: Problem receiving responses from M365 when front light is on

fmenendez, 2019-03-16:
Hello, 

I write you because I am suffering a very strange issue and I would like to ask you if you have suffered it too, my problem is that I have a applicaction that talks with an M365 using the serial port, after struggling a lot with it I have been able to make it work, I mean, I have been able to send messages to the M365 and receive responses. But a problem has suddenly appear when I have turned on the front light, because if this happens I stop receiving responses from the M365, altough I receive messages none of them are response messages, do you know what the problem could be? If so, do you know how this can be fixed?

Thanks a lot in advance for your reply.

Best regards.

camcamfresh, 2019-03-16:
When you turn the headlight on, is the light intensity constant or does it flutter a bit?

Rosch94, 2019-03-16:
Hey @camcamfresh im currently programming my own project. In project everything works fine. Just one thing, if turn the headlight on it begins to flutter a bit. Do you know the reason ?

fmenendez, 2019-03-17:
Hi @camcamfresh,

First of all, thanks a lot for your quick reply.

Answering your question, the light is constant... but the M365 stops replying... 

I give you the Python script I am using to test this issue, maybe you see something wrong in the command or elsewhere:

import serial
import time

while (True):
	ser = serial.Serial();
	ser.baudrate = 115200;
	#ser.port = '/dev/ttyUSB0';
	ser.port = 'COM5';
	ser.parity=serial.PARITY_NONE;
	ser.stopbits=serial.STOPBITS_ONE;
	ser.bytesize=serial.EIGHTBITS;
	ser.timeout=0.05;
	ser.open();
	ser.write("55aa0322011012b7ff".decode('hex'));
	ser.flush();	
	s = ser.read(1024);
	ser.close();
	hex = s.encode('hex');
	if (hex.find("55aa1425") != -1):
		print (hex);
	time.sleep(0.01);		

If I print all the messsages received from M365, instead of only the reply ones as I do in the script, these are the messages I see when I turn on the light:

55aa07206500042629000020ff55aa0920640006262900007200abfe55aa062164000007640009ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa0920640006262900007200abfe55aa062164000007640009ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa0920640006262900007200abfe55aa062164000008640008ff55aa07206500042629000020ff
=================================================
0160004403042629000020ff55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa0920640006262900007200abfe55aa062164000008640008ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa0920640006262900007200abfe55aa062164000008640008ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa0920640006262900007200abfe55aa062164000008640008ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa0920640006262900007200abfe55aa062164000008640008ff55aa07206500042629000020ff
=================================================
55aa07206500042629000020ff55aa07206500042629000020ff
=================================================

As you can see I do not receive any message like:

55aa1453f70306038826c98028d8ff474a3138493741323031301501781ecdfb

which means a reply to me request...

Again, thanks a lot for your help!

pradeepnarava, 2019-03-26:
I need this microcontroller ,how can I get it to India?
Any contacts ?
My mail is pnatrial@gmail.com

camcamfresh, 2019-04-05:
@Rosch94 The headlight's intensity could change for many reasons. My initial guess is something is incorrect with the boost converter; you'll probably want to change the output voltage.

@fmenendez take a look at the message after 55aa09... that is a reply from the motor controller, so it looks like its communicating. What do you mean it stops replying? If there is an error light on the scooter then I would first check your throttle/brake values (perhaps the headlight is changing the power supplied to the hall sensor readings and making the read value invalid). Otherwise, I would check the the message parser/reader. 

@pradeepnarava The microcontroller used is the Particle Electron; you can purchase a development kit from their website at www.particle.io. You will then need to wire the Electron to the scooter. If you're interested I will probably be finishing a box with everything except the Electron that can be mounted to the M365 scooter; I might be interested in selling it later on (probably on eBay or something). I'll post some details in the project.

pradeepnarava, 2019-04-12:
@camcamfresh Hi,thankyou for replying,what is your mail address?have to discuss with you few important points.

pradeepnarava, 2019-04-13:
Can we use Arduino with SIM?
Do you have Arduino code?