camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #23: Is there specific reason to send same command multiple time?

kyuhyong, 2019-04-03:
In void escControl(), I see sending same command repeatedly here.
https://github.com/camcamfresh/Xiaomi-M365-BLE-Controller-Replacement/blob/2dd581ccc5f5c46db6ae40c149a34edba7d66131/MotorControl#L68
Is this for making sure the serial messages not being lost anyway?
Also, why does it need to have motor, and motor1 messages at the same time even though they all look the same.
Thanks,

Kyu

camcamfresh, 2019-04-04:
motor is sent 4 times in an effort to mimic the factory controller. 

motor1 is very similar to motor because it is the same message but also requests the motor controller to send data back to the microcontroller (the x1 structure). 

motor is sent more often than motor1 because the brake and throttle values need to be sent to the motor controller every few milliseconds, whereas we only need replies from the motor controller every few hundred milliseconds.

kyuhyong, 2019-04-05:
Thank you for such a kind explanation.

closed 2019-04-05 by kyuhyong