camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #24: Can't get wheel to spin

ir-fuel, 2019-04-04:
Hello
I have your code running successfully on an Electron, and I can communicate with the M365 and send and receive data. Lock and unlock works for example.

What doesn't work however is trying to spin the wheel. I implemented a loop where every second I increase the throttle value being sent in 100 increments from 685 to 4095, whilst keeping the brake value at 685.

What basically happens is: nothing, until I get into a value range that is too high for the controller and then I get the "invalid throttle value error" (green led blinking in a certain pattern on the motherboard). As soon as I restart the counter from 685 that green led stops blinking until I get too high again.

Is there anything else I need to send in order to get the wheel spinning?

Thanks

camcamfresh, 2019-04-05:
the motor controller has safe guards to prevent the motor from turning on without prior movement. I'm pretty sure this is to prevent the motor from burning out. So for the motor to turn on the wheel has to be turning at a certain rate.

This can be reprogrammed by flashing the motor controller via the microcontroller but I wouldn't want to do it with this microcontroller as any errors could result in a bricked motor controller. I would probably use the factory dashboard to change the motor controller firmware. Check out this link: https://m365.botox.bz/

ir-fuel, 2019-04-05:
Ha, silly me. Of course, just as you can't ride the scooter without kicking it to a certain speed.
Thanks!

closed 2019-07-05 by camcamfresh