camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #26: Status of Replacement Circuit Board

baldemir, 2019-04-08:
Hello,

First of all thank you for your effort on this project. I want to learn details about replacement circuit board that you mentioned that you are working on. I want to learn the progress and estimated cost of the board. I'm interested in your project and would like to buy. We can also discuss on email if you can provide contact to me. 

Best regards

camcamfresh, 2019-04-11:
The first version of the software for the scooter and Android app have been completed (the app is published on the Play Store but is kind of useless without the board). 

I was planning to sell a few prototype boxes for testing later this week, however the circuit broad order had a few errors in the design. I will need to redesign and reorder, which will probably set me back 2 weeks. Haven't thought about price too much, but a guess would be around $100.

mast3rshak3055, 2019-05-15:
Cam please email me directly I have the board I really would like to buy a wiring kit from you

Gskill503@gmail.com

pengiundev, 2019-07-04:
Hello, what's the status of the replacement board? I'm ready to buy a few, can you please email me - ang.angelov@protonmail.com

camcamfresh, 2019-07-05:
I will be posting a link to some boxes on eBay in about a week or two; currently waiting on GPS modules to arrive.

Here are some pics of what it will look like:
https://drive.google.com/drive/folders/1dd2pg6sGPYDqgFMyTAVlMUHgA64pFdBQ?usp=sharing

pengiundev, 2019-07-05:
Will it be possible to build a custom app on top of it, will you publish API details on how to communicate with the box?

camcamfresh, 2019-07-05:
I assume you're talking about a mobile application. It is possible, but I have found it very difficult due to my lack of experience in mobile programming. 

I am using Particle's Mobile Application to send commands and a third-party notification app called Pushover to receive alerts on my phone; this works on Android and iOS. This is my preferred way for the foreseeable future as it very reliable.

I have created an app on the Google Play store called Scooter Connect, however it is not completely reliable. It crashes frequently and if the application is not opened by the user, then it will not monitor for alerts. I can post the variables required on the Particle Electron for the Android Application to interact with the scooter, but I do not want people to rely on it since the Android application is unreliable. Without help from someone with experience in mobile application development, I do not think the custom app will be finished for a few years (as I will not begin studying this subject until later in my college career). 

Particle's API details are here: https://docs.particle.io/reference/device-cloud/api/

pengiundev, 2019-07-05:
Hello, I am willing to do a mobile app as I have a lot of experience as a javascript developer and I can use javascript to develop mobile apps, too.

I can help with the mobile app and I can open-source it, since you've created a great project and it's only honest to contribute back.

If you want to continue on a private conversation, feel free to email me at the email I posted on my first comment.

Can you please tell me what do you want it to have as features?

camcamfresh, 2019-07-06:
It has been a while since I took a look at the code but this is a summary of what I have so far (this is an Android application). There are 5 primary activities: login, device selection, flashing firmware, main activity, and a control activity. There is also one background service.

Login activity takes the users Particle email and password and retrieves an access token from the Cloud.

Directly after logging in, one is directed to the device selection activity, where they can select an online  Particle Electron. Once a device is selected, the application starts a background service that monitor for events of that device - primarily GPS coordinates and alarm alerts.

If the Electron's firmware doesn't have the correct version number, the user is directed to the flash firmware activity, where they can upload pre-compiled compatible firmware to the Electron. The user is then directed back to the main activity.

The main activity is a Google map with the user's current location. If the user is logged in, they will be able to see the logout button and device selection button. If a device is selected, a dashboard will appear on the bottom of the screen with an lock/unlock button, scooter battery level icon, and a button to the control activity. In addition to the dashboard, if a device is selected, two additional buttons will appear under the 'center to user location' button; one to center to the scooter's location and the other sounds the scooter's alarm.

The control activity is a split screen. The top half contains statistics about the scooter; the bottom half is composed of toggle switches that control the scooter (unlock/lock, headlight on/off, eco mode on/off etc.).

The background service starts when a device is selected and stops when the device selection page is opened. When it receives gps data, the service saves the data in an sql lite database stored on the phone. If the main activity is opened and the scooter's location is not available the map will show the last location of the scooter with a different color marker. When it receives an alarm alert, it sends a notification to the user.

I am currently having problems keep the service running when the application is closed. 
I also want to show a timestamp when a user presses the scooter's marker on the map. I also would like to give the user the ability to scroll through past GPS locations along with the matching timestamps.

camcamfresh, 2019-07-16:
I currently have two boards for sell on eBay if anyone is still interested.
https://www.ebay.com/itm/163765642620

sproutbrussell, 2019-08-25:
any more for sale?

robbiet480, 2020-03-04:
@camcamfresh Are you planning to put any more on eBay?