camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #28: Help Finding Connectors for Throttle and Brk

camcamfresh, 2019-04-11:
This isn't directly code related, but I have been searching for weeks to find the same type of connectors that Xiaomi and Seagway use on their controllers with no luck.

I cannot find the 3-pin, 1.5mm female JST connector used for the brake & throttle sensors. I'm thinking I will have the same difficulty searching for the 2-pin 1.5mm female JST connector for the headlight.

[image]

I have been able to find the male 3-pin JST connector & the cylindrical 4-pin connector, but cannot find the other connectors. Any help in finding these connectors would be appreciated.

closed 2019-07-05 by camcamfresh

aleixdetoro, 2019-10-01:
@camcamfresh Where did you find those connectors? Im struggling with it (specially the cylindrical)... Thanks!

Ursescu, 2019-10-11:
@camcamfresh Hey, can you share the name of the JST connectors ? Are those JST SH ?

andrescli, 2020-09-25:
What is name of the 1.5mm JST throttle connector

andrescli, 2020-09-25:
Please can anyone share the name of the 1.5mm JST throttle connector??

rontav, 2020-10-02:
The connectors used for throttle, brake and the led light are: JST ZHR-3, JST ZMR-03 and JST ZMR-02.
Note that these are the connectors that mate with the ones on the handlebar, aka the connectors used for the dashboard.

Ursescu, 2020-10-02:
The cylindrical one can be found on Aliexpress under the name **Julet Connector**. The original ones I think are called Higo connectors from A&C solutions. [(Here)](https://www.higoconnector.com/products/z209bg-p-00-a0-1000/0999K#title).