camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #29: Renting Project

zstergios, 2019-05-13:
Hi,
I'm web programmer and 2 years I have spent some time with arduino and electronics.
Your project it's lifesaver. My client wants to rent the scooter like LIME etc.

I can build the APP, the communication api between CONTROLLER->SERVER->MOBILE and backend for management.

I know that there are chinese options out there that I'm thinking it's worth tring to make a demo controller my own.

Still I don't have any Xiami scooter to my hands to able to digg but I'm making a research to be prepared of what I will face.

**So, let's start, please help if the following are right conclusion or wrong**

1)  There 2 power lines 5v & 48v (or 36v), so I don't need regulator to 5v

2) As I read here, the board controller has some pins for output (TX/RX etc) so arduino can write/read via serial

3) firmware.ino seems that I can get/set readings for almost all common renting combabilities

4) Xiami has his own Bluetooth connection. As I know only registered devices (owner) can connect
If not I should disable it.

5) Has some space near to the board controller to add my custom PCB

6) 5V line has over 2 Amps current

Some considerations
I would like to use something that I have worked before like arduino. But has few space, GPS+GSM+ code may it will not have enough ram to do the POST requests.
The other solution is to create a PCB with 2 arduino (bear bones) to double the space  & ram (more tricky)

closed 2019-05-18 by zstergios

pilotdeveloper, 2019-05-18:
The Bluetooth is not secure at all and you should absolutely disable that. With that, one could easily replace your main board and walk with your scooter.

Kenner-G, 2019-06-17:
Hello, how did the project go?