camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #30: Contributor access?

pilotdeveloper, 2019-05-13:
Mind adding me as a contributor?  :)

camcamfresh, 2019-06-23:
Not too knowledgeable about GitHub, but I think changed the settings. Let me know if it worked.

NitroProp, 2019-09-03:
Hi, i would like to contribute by design a new pcb for the project and maybe a new case design. if you want and if there is no difficulty for you we can change the electron with this : https://store.particle.io/collections/cellular/products/e-series-global-3g-module that is more pcb friendly. Let me know if for you is okey 
Regards

camcamfresh, 2019-09-06:
I see, it looks good. I use the Electron, so that I can use it other project too (it's really easy to remove with a female header socket). If you integrated this onto the circuit board you wouldn't be able to remove it, but neither would other people. Do you have a picture of the design?

NitroProp, 2019-09-09:
Well right now i'm replicating the board with the Electron but instead of a standard placement on the board, there will be an hole on the pcb shaped like the electron and on the edges there will be solder pads so the board will sit flush with the surface but everithing if thinner and you can acees the sim on the bottom. I've done my best to explain that XD I will keep you updated !

NitroProp, 2020-01-17:
Almost completed the pcb design:
https://drive.google.com/file/d/1tlcTDExb_Qfaydw-JnIANXQ0agfvYybo/view?usp=sharing
waiting to get the other module, in my region is hard to get at the moment, hope you like the pcb design