camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #31: Anyone encountered sudden voltage drop of 5V from scooter?

kyuhyong, 2019-06-18:
Sorry, this issue has nothing to do with the code itself but I want to share.
I made a board which carries a particle electron board and interface with the scooter.
Main 5V Power is mainly supplied directly from the scooter and has been working although they are not exactly 5V and roughly around 4.8~4.9V measured from multimeter.

The problem is this voltage is not consistent and it can be dropped as low as 4.2V sometime. 
This causes  problem maintaining connection with the user.

Has anyone encountered such problem? If so, how did you solve this problem?

camcamfresh, 2019-06-18:
I think I had a problem similar to this. What does the Electron do when you try powering it from only the scooter (without the Electron battery)?

kyuhyong, 2019-06-19:
Thank you for your comment. I found the board won't work if the voltage goes below 4.3V if not powered with the electron battery.
Seems like the regulator on the scooter is limited.

camcamfresh, 2019-06-19:
I had a power issue from the motor controller a while back, but I'm not sure if it's the same problem.

From what I remember, when the Electron battery was not connected the Particle Electron wouldn't connect to a cell tower on start up and the led would pulse white slowly. 
If the Electron battery was plugged in, the Electron would connect, but not charge; once the battery was drained enough the Electron led would show it was connected to the cell tower but couldn't actually receive commands through the cloud.

I eventually found a Russian forums page that helped me solve this problem. I had to take out the motor controller, discharge the motor controller (it has capacitors), and solder over a 0 ohms resistor. Take a look at the picture.

[image]

pilotdeveloper, 2019-06-19:
Please note that touching those capacitors can kill you, so make sure it's
discharged!

camcamfresh, 2019-07-05:
@kyuhyong any updates?

kyuhyong, 2019-07-05:
@camcamfresh  Sorry for late reply.
I found the voltage regulator is quite small and highly doubt if it can make over 250mA. 
So I decided to put another voltage regulator which can make 5V from 12V in the board as shown below picture.
[image]

This solved my issue and have not experienced any problem with power.

closed 2019-07-05 by camcamfresh

mossafaei, 2019-08-28:
@kyuhyong  @camcamfresh Hi, Can you show me where is the GND for 12v on the motherboard? This is not clear in the above photo, i can't find it.

camcamfresh, 2019-09-06:
I rarely go into the motor controller's motherboard, but I would assume it is the same as the 5V & 40V GND.
Personally, I would try doing it another way since the capacitors can kill when charged & loose connections happen.

kyuhyong, 2019-10-28:
> @kyuhyong @camcamfresh Hi, Can you show me where is the GND for 12v on the motherboard? This is not clear in the above photo, i can't find it.

All GND is connected with copper ground and to the battery GND as well, so finding is easy if you have any multi-meter at hand.