camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #33: Hardware setup throttle resistors info

pengiundev, 2019-07-17:
Hello, can someone explain why we need the second resistor? 

> Connect Throttle Input Wire to 1.6K Resistor and the other side of the resistor to Particle Electron's A4 Pin.
> Connect a 3.3K Resistor to the Ground Line and the other side of the resistor to the Particle Electron A4 Pin.
>

camcamfresh, 2019-07-18:
The second resistor completes a voltage divider. The brake/throttle inputs can deliver up to 5V (really 4.9V) when fully pressed; the Electron only reads up to 3.3V (even though the pins are 5V tolerant).

You don't have to implement the voltage divider, but once the input delivers 3.3V or more it won't matter how much further you pull the throttle (or brake) because it will be read at full speed anyways. 
From what I remember when testing this, 3.3V delivers an analog read of 4095 on the Electron as does 4.9V. Thinking about it, the Electron might actually read up 4V, but either way your losing some of the range when pressing the brake/throttle sensors.

So if you don't implement the voltage divider, your brake and throttle inputs will only work up to the 3.3V (or 4V) which equates to pulling the lever roughly 2/3 (or 4/5) of it's possible range.

pengiundev, 2019-07-19:
Thank you for the very detailed explanation! It's perfectly clear now :)

closed 2019-07-19 by pengiundev