camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #34: How to keep particle on when we power off scooter

arminsalcin, 2019-07-24:
Hello to all , 

So my question is when you turn off scooter by particle the 5V line is 0V so particle is off how can you get command from cloud to turn on scooter when 5v line is off ?

mossafaei, 2019-08-07:
You need to use battery when scooter is off

arminsalcin, 2019-08-07:
Thank yoj for answerinf , also question how to keep scooter always in unlocked state ?

mossafaei, 2019-08-09:
https://github.com/camcamfresh/Xiaomi-M365-BLE-Controller-Replacement/issues/19

arminsalcin, 2019-08-10:
Thank you for answering , i already read that but what if scooter is riding and i lock and unlock it by time it will crash. It is little bit risky ...

mossafaei, 2019-08-10:
You can get the state of wheels and checking if scooter is riding or not

arminsalcin, 2019-08-10:
Ok thank you for helping , i figure it out !

closed 2019-08-10 by arminsalcin