camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #36: Throttle stops working for a second randomly while driving

pengiundev, 2019-08-07:
Hey, have you encountered the throttle suddenly stopping to work while driving? It stops working for a second randomly and I'm not sure why that happens. 

I think that it may be related to particle electron disconnecting and then reconnecting but from looking at the source code, the throttle should work regardless of being connected to the internet or not?

pengiundev, 2019-08-08:
It happens in the event of disconnect, shortly before or after the particle reconnects again

pengiundev, 2019-08-09:
OK, I found where the issue is, I'm still testing but it's working for now. Is it OK to do a PR with the fix?

camcamfresh, 2019-09-06:
Sounds good! I'm going to try and put the old firmware version for future reference.
I'm not sure if there is anything for the PR I need to do, so let me know if I do.