camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #37: Very strange communication problem. M365

Fcatao1978, 2019-08-21:
Hi guys! I have a real strange battery/ comunication problem. My Battery is almost fully charged and it shows in the app as 3% forever. Even when I go for a ride in the scooter (it reaches only slow speed) the app still show 3%.  I am attaching some pics, is this a board problem? Why my battery blinks a red led when the scooter off? I am new here so I don’t know if this post is in the correct place, sorry.

[image]
[image]
[image]

Fcatao1978, 2019-08-21:
[image]
[image]
[image]
[image]

Fcatao1978, 2019-08-21:
I changed the circuit board but the problem continues.

camcamfresh, 2019-09-06:
Which circuit board did you change?
My best guess right now is that something is wrong with the BMS; there is a button on the BMS circuit board (within the plastic), that I believe resets the BMS. I would try that.