camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #39: M365 connectors

Ursescu, 2019-10-12:
Hey, can you share the name of the JST connectors and the name for the blue connector (4pins)? It can be very useful if someone wants to implement a dashboard replacement.

pengiundev, 2019-10-13:
On the connectors it says JST ZH and XH but I can't find similar looking connectors. Especially cable connectors, not board connectors.

Ursescu, 2019-10-13:
I want to know the name of connectors fort the brake throttle and heading so I can create a plug and play board.

sproutbrussell, 2019-10-16:
Not 100% sure but I think they're 1.5mm pitch, which would make them JST ZH.  JST ZH are also compatible with JST ZR and JST ZM.  Again, not 100% sure, but the trapezoidal 2/3-pin female JSTs on the throttle, brake, and headlight looks like JST ZM:  

http://www.jst-mfg.com/product/pdf/eng/eZM.pdf?5da6ba2363b7d
https://en.wikipedia.org/wiki/JST_connector

camcamfresh, 2019-10-17:
Here is where I got the JST connectors: https://order.jst-mfg.com/InternetShop/app/index.php
SAN-002T-0.8A
6P-SAN
4P-SAN
SZH-003T-P0.5
ZHR-3 (PP)
ZMR-02
ZMR-03

I'd need to look for the other connector, idk if I'd be able to find it again.

Honestly, it would probably be easier buying a cheap board and using the connectors on that board (which would require de-soldering and some patience) if you just want a board for yourself

Ursescu, 2019-10-17:
Awesome.

Thank you for your time and information. 

For the blue round one, I think it's called Higo connector. But it's annoying that I can't find where to buy them. (some provider link would be awesome).

joeybab3, 2020-06-23:
> Awesome.
> 
> Thank you for your time and information.
> 
> For the blue round one, I think it's called Higo connector. But it's annoying that I can't find where to buy them. (some provider link would be awesome).

This is an old thread that I stumbled on from google, the round connector is more traditionally known as a "julet" connector and sold under that name or as an "e-bike" connector, but a search for Julet connector should yield the one your looking for.

Ursescu, 2020-06-26:
Closing this issue.

closed 2020-06-26 by Ursescu