camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #40: GPS problem: Arduino.h No such file or directory

aleixdetoro, 2019-10-16:
I am trying to run the old_firmware_v6e.ino and procedural_firmware_v9e.ino for using the GPS, the basic_usage.ino already works with my scooter but wihtout the GPS.

When I am adding the NEOGPS Library (including the NMEAsimple.ino, as what I have read in one earlier issue: https://github.com/camcamfresh/Xiaomi-M365-BLE-Controller-Replacement/issues/8) and trying to verify the code, it gives me the following error: 

**lib/NeoGPS/NMEAGPS.h:24:21: Arduino.h: No such file or directory**

Did you have the same problem?

Last question, what is exactly the difference between the basic_usage and the other firmwares available apart from the GPS?

Thanks!

camcamfresh, 2019-10-17:
Basic usage was to show how to use the M365 library, but I was having problems with that and haven't had time to look into what was wrong. 
I uploaded the old_firmware_v6e.ino, because that was a version that I remember having no known bugs, but the code is kind of messy. The procedural_firmware_v9e.ino is (supposed) to be a cleaner version of v6e, but there are still a few parts that need work (e.g. the headlight stays on when lock).

I will try adding clearer documentation on each file when I have time. 

As for the NeoGPS Library, I included the entire library. I'm not sure if the NMEAsimple.ino example includes all of the files. I'd try removing all of the libraries, searching for "NeoGPS" library through Particle's IDE, and click include in project instead of using one of the examples. Let me know how it goes.

adezel, 2019-10-17:
Thanks Cameron! 
Regarding the GPS it hasn't worked. So I have the old_firmware_v6e.ino code,  then I add the NEOGPS library and change from the old_firmware_v6e.ino the "include NeoGPS.h" to "include NMEAGPS.h". It seems in the NEOGPS library the NeoGPS.h does not exist, thats why I have to do that change.

So I get the following error when changing this:
**lib/NeoGPS/NMEAGPS.h:24:21: Arduino.h: No such file or directory**

How did you do it?

By the way can this code work with the Adafruit GPS? Have you tried it adding the ADAFRUIT GPS Library?

camcamfresh, 2019-10-22:
I go to build.particle.io & login; then you would want to create a new app and name it. Before you copy any of the code, I would make sure you select your Electron Particle device under the "Devices" tab. This way when you "Verify" (compile) and "Flash" the code, the IDE knows that you are using the Electron; this way we can #include "Serial5/Serial5.h" without any errors. 

Next I would search for the NeoGPS library; it is about 6.8K. Select the library and "Include in Project". This will ask you which project you want to include the library in. Select the app name in the first step and select confirm. This will cause a statement: "#include <NeoGPS.h>" to be added your app. Change this to "#include <NMEAGPS.h>". Verify the your app and make sure it compiles. If successful then you can copy the code from github into your app.

I have only tried using the Neo GPS; I wouldn't see a problem with using the Adafruit GPS assuming there is a Adafruit library for the GPS.

adezel, 2019-11-15:
It finnally worked! 

Now the problem appears when getting the FIX from the GPS Module. I have connected the GPS to the 3.3V, GND, C0 and C1, with all the other parts on the PCB , and even if I put the gps outside it does not get a fix position (led blinking).

However when I try just the electron and the GPS module connected with cables, the FIX GPS position is received very fast.

Have you had any trouble with this? 

I have read that puting some cap in the VIN-GND of the GPS could solve the problem...