camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #44: Arduino Nano 33 BLE - Can't send commands

bartoszwarszawa, 2020-02-21:
I am using your procedural_firmware_v9e.ino for reference to make an Arduino Version of this with SIM808 GSM chip aaand the commands don't really work. 
I did it the same as you. I am translating the message from the DRV controller put them in stats structure and at the end of the translate function check if the commands are triggered.
Commands are in a command structure as well and even if i hardcode (change) the commands from the default ones when turning on the scooter it doesn't change.
I made a custom function to view the protocol messages comming from DRV controller and i can see that the changes are not triggered. And also when i try to turn on the tail light it doesn't trigger as well. Any idea on fixing it? The code is more or less the same as yours.

bartoszwarszawa, 2020-02-21:
I did downgrade the DRV firmware version to 1.3.4 as in the protocol listed.

technoo10201, 2020-03-01:
Hi @bartoszwarszawa 
Where did you find the Timer library ? I'm not able to find it...
Thanks in advance !