camcamfresh/Xiaomi-M365-BLE-Controller-Replacement #46: Is it possible to change max speed using basic_usage.ino?

SKajiwara, 2020-03-07:


bartoszwarszawa, 2020-03-10:
You need to change engine driver with M365 DownG - You need to use original BLE controller for that. There are firmware generators when you can lift engine electric limits. Btw you don't want to do that if you don't know what you're doing. Those engines heat up pretty fast when you exceed 250W on them.

SKajiwara, 2020-03-10:
Thank you for reply.
I see what you mean. I didn't know about the engine.

I am not trying to fasten the scooter too much. 
I wonder if I can lower the speed even by subtracting input values from Throttle to a lower value. What do you think?

bartoszwarszawa, 2020-03-11:
You will loose throttle controll. (By default you have 0-100) if you substract you will press the throttle a little bit and you will not have any input send to the engine. It's better to place higher electric limit on the engine instead because the engine will lower the power of itself and also you will not have issues when  going uphill. (That's the scenario when the engine is doing 2-3x the power and it's heating pretty heavily)

SKajiwara, 2020-03-11:
Ok thank you. I'm not knowledgeable enough to change the electric limit on the motor.

Do you think multiplying 0.8 to throttle can limit speed at 80% of its default max(15.5 miles/hour)??